dvc94ch/pykicad #2: rotation component of `at` isn't bubbled up

wez, 2017-04-15:
There are a couple of instances of:

```
                          Tag.parser('at', number, number,
                                    # Bug in some package definitions?
                                    Suppress(Optional(number))) &
```

I'd recommend removing the `Suppress` so that the rotation is bubbled up as an optional 3rd element of the `at` tuple.

dvc94ch, 2017-04-15:
Can you explain what it's for? Kicad deals with 2D geometry right?

Thanks!

dvc94ch, 2017-04-15:
Says in the title sorry

wez, 2017-04-15:
Say you have a rectangular pad; in the footprint editor you can set it to be rotated so that it renders as a diamond rather than a square.  This rotation is stored in the 3rd element of the `at` property.

dvc94ch, 2017-04-19:
Fixed

closed 2017-04-19 by dvc94ch