dvc94ch/pykicad #13: gr_line support

wez, 2017-04-23:
I was using the pcbnew function `DRAWSEGMENT` for this, so I blindly assumed that this could use the `Segment` type for this, but on closer inspection, this API was creating a series of `gr_line` elements that are almost the same properties as `Segment` but don't have an associated net.

I'm using these on the `Edge.Cuts` layer to shape my board.

I took a stab at adding support for this; it appears to do the right thing for my design; does this look right?

Note that the `start` and `end` elements are required to be first.

```
diff --git a/pykicad/pcb.py b/pykicad/pcb.py
index 2718f6a..0ca353e 100644
--- a/pykicad/pcb.py
+++ b/pykicad/pcb.py
@@ -20,6 +20,27 @@ class Segment(AST):
                                       layer=layer, net=net, tstamp=tstamp,
                                       status=status)

+class Line(AST):
+    tag = 'gr_line'
+    schema = {
+        '0': {
+            '_tag': 'start',
+            '_parser': number + number
+        },
+        '1': {
+            '_tag': 'end',
+            '_parser': number + number,
+        },
+        'width': number,
+        'layer': text,
+        'tstamp': hex,
+    }
+
+    def __init__(self, start, end, width=None, layer='F.Cu',
+                 tstamp=None):
+        super(Line, self).__init__(start=start, end=end, width=width,
+                                   layer=layer, tstamp=tstamp)
+

 class Via(AST):
     tag = 'via'
@@ -69,10 +90,15 @@ class Pcb(AST):
         'vias': {
             '_parser': Via,
             '_multiple': True
-        }
+        },
+        'lines': {
+            '_parser': Line,
+            '_multiple': True
+        },
     }

     def __init__(self, version=1, host=['pykicad', 'x.x.x'], nets=[], modules=[],
-                 segments=[], vias=[]):
+                 segments=[], vias=[], lines=[]):
         super(Pcb, self).__init__(version=version, host=host, nets=nets,
-                                  modules=modules, segments=segments, vias=vias)
+                                  modules=modules, segments=segments, vias=vias,
+                                  lines=lines)
```

wez, 2017-04-24:
Just noticed that this is tracked as a bullet in #9

dvc94ch, 2017-04-24:
Looks good!

dvc94ch, 2017-07-10:
Done

closed 2017-07-10 by dvc94ch