dvc94ch/pykicad #14: zone support

wez, 2017-04-23:
Another API I was using in pcbnew was `InsertArea`.

Here's a slightly abbreviated dump of one of the areas that I'd created; I've removed some of the points from the polygon:

```
  (zone (net 2) (net_name GND) (layer F.Cu) (tstamp 58FCDE74) (hatch edge 0.508)
    (connect_pads (clearance 0.508))
    (min_thickness 0.254)
    (fill (arc_segments 16) (thermal_gap 0.508) (thermal_bridge_width 0.508))
    (polygon
      (pts
        (xy 67.944355 25) (xy 67.944355 25) 
      )
    )
  )
```

wez, 2017-04-23:
(this is useful for ground and power planes)

wez, 2017-04-24:
I don't know how much value this adds over the bullet in #9

dvc94ch, 2017-07-12:
Fixed

closed 2017-07-12 by dvc94ch