dvc94ch/pykicad #15: symbolic helpers

wez, 2017-04-23:
Following on from some of the conversation in #7, I found myself needing these two helper functions today:

```
def set_reference(module, name):
    ''' Change the reference/identifier for a freshly loaded module.
        Aside from changing the name, we also need to update the
        textual elements, otherwise they are left at the default.
        Some components default to names like '>NAME' which
        causes some compatibility problems with FreeRouter '''
    module.name = name
    for text in module.texts:
         if text.type == 'reference':
              text.text = name
```

Ideally would be a `set_name` or `set_reference` method or similar on the `Module` class.

```
def pad_by_name(module, name):
    ''' Returns a named/numbered pad.
        The pads in the list may be in an arbitrary order, or be non-consecutive.
        Ideally, this would be a method of the Pad class '''
    for pad in module.pads:
         if pad.name == name:
              return pad
    return None # or could raise an IndexError or KeyError
```

This latter one is particularly helpful when you have a footprint with many pins; the examples in the readme work out ok because it doesn't matter which end of a resistor you get when you index into the pads list :)

Thoughts on making these first class citizens?

dvc94ch, 2017-04-24:
Looks good! Would you like to submit a PR?

dvc94ch, 2017-04-25:
Actually on second thought, I think that we should leave the Module class as is.

I'm thinking about a Footprint class that represents an instance of a module. This saves dealing with unnecessary information.

This footprint class would contain a shapely Polygon representing the courtyard, a set of points representing the pads, a reference, a link to it's module and nicer support for connecting nets and segments and pad indexing by name. This also allows keeping shapely and numpy as optional dependencies.

dvc94ch, 2017-07-10:
Done

closed 2017-07-10 by dvc94ch