dvc94ch/pykicad #20: Can't parse a kicad_pcb file

santisaran, 2018-07-16:
I'm trying to parse a kicad_pcb file:

```
from pykicad import pcb 
pcbk = pcb.Pcb.from_file("example.kicad_pcb")
```
here is the kicad [example.kicad_pcb](https://drive.google.com/open?id=1dwVgdm7YI0gcvSSlT_XIuK9fbJoABxDV) file

I get this error in the output, with this file and with any other kicad_pcb file:

---------------------------------------------------------------------------
ParseException                            Traceback (most recent call last)
<ipython-input-15-9abaeb0085ee> in <module>()
----> 1 pcbk = pcb.Pcb.from_file("e:/example.kicad_pcb")

C:\ProgramData\Anaconda3\lib\site-packages\pykicad\pcb.py in from_file(cls, path)
    789     @classmethod
    790     def from_file(cls, path):
--> 791         return Pcb.parse(open(path, encoding='utf-8').read())

C:\ProgramData\Anaconda3\lib\site-packages\pykicad\sexpr.py in parse(cls, string)
    428             cls._parser = generate_parser(cls.tag, cls.schema)
    429 
--> 430         parse_result = cls._parser.parseString(string)
    431 
    432         result = {}

C:\ProgramData\Anaconda3\lib\site-packages\pyparsing.py in parseString(self, instring, parseAll)
   1214             else:
   1215                 # catch and re-raise exception from here, clears out pyparsing internal stack trace
-> 1216                 raise exc
   1217         else:
   1218             return tokens

C:\ProgramData\Anaconda3\lib\site-packages\pyparsing.py in parseString(self, instring, parseAll)
   1204             instring = instring.expandtabs()
   1205         try:
-> 1206             loc, tokens = self._parse( instring, 0 )
   1207             if parseAll:
   1208                 loc = self.preParse( instring, loc )

C:\ProgramData\Anaconda3\lib\site-packages\pyparsing.py in _parseNoCache(self, instring, loc, doActions, callPreParse)
   1070             if self.mayIndexError or loc >= len(instring):
   1071                 try:
-> 1072                     loc,tokens = self.parseImpl( instring, preloc, doActions )
   1073                 except IndexError:
   1074                     raise ParseException( instring, len(instring), self.errmsg, self )

C:\ProgramData\Anaconda3\lib\site-packages\pyparsing.py in parseImpl(self, instring, loc, doActions)
   2542                     raise ParseSyntaxException( ParseException(instring, len(instring), self.errmsg, self) )
   2543             else:
-> 2544                 loc, exprtokens = e._parse( instring, loc, doActions )
   2545             if exprtokens or exprtokens.haskeys():
   2546                 resultlist += exprtokens

C:\ProgramData\Anaconda3\lib\site-packages\pyparsing.py in _parseNoCache(self, instring, loc, doActions, callPreParse)
   1070             if self.mayIndexError or loc >= len(instring):
   1071                 try:
-> 1072                     loc,tokens = self.parseImpl( instring, preloc, doActions )
   1073                 except IndexError:
   1074                     raise ParseException( instring, len(instring), self.errmsg, self )

C:\ProgramData\Anaconda3\lib\site-packages\pyparsing.py in parseImpl(self, instring, loc, doActions)
   2742         if tmpReqd:
   2743             missing = ", ".join(_ustr(e) for e in tmpReqd)
-> 2744             raise ParseException(instring,loc,"Missing one or more required elements (%s)" % missing )
   2745 
   2746         # add any unmatched Optionals, in case they have default values defined

ParseException: Missing one or more required elements (Group:({Suppress:("(") Suppress:("net_class") Group:({string enclosed in double quotes | W:(0123...)}) Group:({string enclosed in double quotes | W:(0123...)}) {[Group:({Suppress:("(") Suppress:("clearance") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("trace_width") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("via_dia") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("via_drill") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("uvia_dia") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("uvia_drill") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("diff_pair_width") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("diff_pair_gap") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})]} [Group:({Suppress:("(") Suppress:("add_net") {string enclosed in double quotes | W:(0123...)} Suppress:(")")})]... Suppress:(")")}), Group:({Suppress:("(") Suppress:("module") Group:({string enclosed in double quotes | W:(0123...)}) {[Group:({Suppress:("(") Suppress:("version") W:(0123...) Suppress:(")")})] & [Group:("locked")] & [Group:("placed")] & [Group:({Suppress:("(") Suppress:("layer") {"F.Cu" | "B.Cu"} Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("tedit") W:(0123...) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("tstamp") W:(0123...) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("at") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) [Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]})] Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("descr") {string enclosed in double quotes | W:(0123...)} Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("tags") {string enclosed in double quotes | W:(0123...)} Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("path") {string enclosed in double quotes | W:(0123...)} Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("attr") ["smd"] ["virtual"] Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("autoplace_cost90") W:(0123...) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("autoplace_cost180") W:(0123...) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("solder_mask_margin") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("solder_paste_margin") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("solder_paste_ratio") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("clearance") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("zone_connect") W:(0123...) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("thermal_width") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("thermal_gap") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})]} {[Group:({Suppress:("(") Suppress:("fp_text") Group:({"reference" | "value" | "user"}) Group:({string enclosed in double quotes | W:(0123...)}) Group:({Suppress:("(") Suppress:("at") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) [Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]})] Suppress:(")")}) {[Group:({Suppress:("(") Suppress:("layer") {string enclosed in double quotes | W:(0123...)} Suppress:(")")})] & [{Suppress:("(") Suppress:("effects") Empty {[{Suppress:("(") Suppress:("font") Empty {[Group:({Suppress:("(") Suppress:("size") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:({Suppress:("(") Suppress:("thickness") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})] & [Group:("bold")] & [Group:("italic")]} Empty Suppress:(")")}] & [Group:({Suppress:("(") Suppress:("justify") {"left" | "right" | "top" | "bottom" | "mirror"} Suppress:(")")})] & [Group:("hide")]} Empty Suppress:(")")}] & [Group:("hide")] & [Group:({Suppress:("(") Suppress:("tstamp") W:(0123...) Suppress:(")")})]} Empty Suppress:(")")})]... & [Group:({Suppress:("(") Suppress:("fp_line") Group:({Suppress:("(") Suppress:("start") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")}) Group:({Suppress:("(") Suppress:("end") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")}) {[Group:({Suppress:("(") Suppress:("layer") {string enclosed in double quotes | W:(0123...)} Suppress:(")")})] & [Group:({Suppress [... cut ...] 123...)}]}) Suppress:(")")})]... Empty Empty Suppress:(")")}] Empty Suppress:(")")}] & [{Suppress:("(") Suppress:("arrow2b") Empty [{Suppress:("(") Suppress:("pts") [Group:({Suppress:("(xy") Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Combine:({["-"] W:(0123...) [{W:(.) W:(0123...)}]}) Suppress:(")")})]... Empty Empty Suppress:(")")}] Empty Suppress:(")")}] & [Group:({Suppress:("(") Suppress:("tstamp") W:(0123...) Suppress:(")")})]} Empty Suppress:(")")})) (at char 712), (line:39, col:3)

choltz95, 2019-01-18:
Yes - can't even parse the demo files that kicad downloads with.

zfphil, 2019-04-06:
My branch may have fixed this - just created a PR. Seems like the S-expression schema was just missing a few keys, and if you have multiple user-defined trace widths it will still fail (working on that).